fkanehiro/choreonoid_ros_pkg #1: プラグインのインストール先

fkanehiro, 2015-10-20:
@yosuke さん、
Choreonoidがすでにインストールされている環境で、インストールの手順にしたがってインストールを行うとプラグインは既存のChoreonoidの方にはいってしまうようです。
これは想定通りでしょうか。そうだとしてdevelの下のChoreonoidのビルドは省略できるでしょうか。

fkanehiro, 2015-10-20:
プラグインが想定外のところに入っていることが原因かもしれないのでこちらに書いておくと、プラグインが入った方のChoreonoidを引数なしで起動し、何もせずに終了すると次のエラーメッセージとともにコアダンプします。
```
choreonoid: /usr/include/boost/smart_ptr/shared_ptr.hpp:653: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = ros::NodeHandle; typename boost::detail::sp_member_access<T>::type = ros::NodeHandle*]: アサーション `px != 0' に失敗しました。
中止 (コアダンプ)
```
また、チュートリアルにあるサンプルプロジェクトを引数に起動するとシミュレーション開始時にコアダンプします。
（gdbでスタックトレースを見てみましたが、使えそうな情報はありませんでした）

yosuke, 2015-10-20:
情報ありがとうございました。

pkg-configからインストールフォルダを取得しているのですが、catkinの外のpkg-configを見てしまっているようなのでそれを直します。

コアダンプもおそらくプラグインのデストラクタ処理（roscoreへの接続関連）がきれいにできていないせいですね。
直します。

yosuke, 2015-10-26:
catkinコマンドを使う限り（＆私の環境）ではdevel以下に入ってくれるようです。なぜそうなっているのか、もう少し調べてみます

coreダンプの理由も探っているのですが、おそらく継承元クラスのvirtual関数が足りないという理由でRosWorldItemが作れなくなっている（最近のChoreonoidのアップデートの影響？）ので、ここをまずは直します。

yosuke, 2015-10-26:
```
source devel/setup.bash
```
をやっていないせいかもしれません。
ドキュメントに足しておきます。

fkanehiro, 2015-10-26:
#4 の修正でdevel以下にプラグインがインストールされることを確認しました。
また、 シミュレーションをスタートすると落ちる問題も起きないことを確認しました。
(choreonoidを引数なしで立ち上げて落とすとアサーションが起きるのは相変わらずです）
次なる問題として、JVRC-1が自由落下するだけなのにシミュレーションが非常に遅い、ということが判明しましたので別issueにしておきます。

closed 2015-10-26 by fkanehiro