fkanehiro/choreonoid_ros_pkg #6: シミュレーションが遅い

fkanehiro, 2015-10-26:
チュートリアルにあるJVRC-1のシミュレーション（JVRC-1が自由落下するシミュレーション）を行うと、３倍くらい実時間に比べて遅くなります。ＢｏｄｙＲＯＳ、ＷｏｒｌｄＲＯＳのアイテムがなければ２倍速いです。

yosuke, 2015-10-26:
シミュレーションループの中で関節情報などを毎回生真面目にpublishしてしまっているからかもしれないですね。

原因を探ってみます。

終了時のcoreダンプの件はデストラクタ処理の不足で間違いなさそうです。

yosuke, 2015-10-26:
BodyROSが遅い原因（BodyROSを消すと速い速度に戻る、WorldROSは変わらない）というところまでわかりました。

PA10にBodyROSを追加した時にはそこまで遅くなかったので、センサ系のpublishが良くないのだと思います。

調整します。

yosuke, 2015-10-26:
センサ系を更新された時のみpublishすることでだいぶん早くなりました。

joint_stateに関しても全てのループでpublishするわけではないようなので、そこも調整します。
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_joint_state_publisher.cpp#L66

fkanehiro, 2015-10-27:
#7 の変更で速くなりましたが、まだ1.7倍くらいかかります。
GLVisionSimulatorアイテムを削除すると0.03倍とかになるのですが、画像まわりはもっとスピードアップできないでしょうか。

fkanehiro, 2015-10-27:
ROSがよくわかってませんが、publishすると通信が発生する？それともバッファに書き込むだけで実際にはspinOnceで通信発生？subscribeしている人がいなくても通信が発生？
いずれにせよ、シミュレータのメインループで通信が発生するのは問題だと思います。

yosuke, 2015-10-27:
情報ありがとうございます。画像はピクセルのコピーのところがダメな実装になっていそうなので、ポインタ＆memcopyを使って書きなおして見ます。

ROSはsubscribeしている人がいなくても通信が発生してしまうようで、その場合に通信をしないようにする実装をユーザ側でしているのを見た記憶があります。
実際の通信はspinのタイミングで行われていると私は理解しているのですが、今の実装だと確かにループの中でspinしてしまっています。gazebo pluginについては以下の部分で非同期でspinがかかっていますので、それと同じように実装したいと思います。
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_ros/src/gazebo_ros_api_plugin.cpp#L141

yosuke, 2015-10-27:
gazebo pluginについてもvisionについては、subscriberの数を見てpublishの制御をかけていますね。
これも同じ実装にしたいと思います。

https://github.com/ros-simulation/gazebo_ros_pkgs/blob/534470d4e6375519f4508dac47d789c5b8b9acdf/gazebo_plugins/src/gazebo_ros_camera.cpp#L89

yosuke, 2015-10-27:
ただ私の環境だと、GLVisionSimulatorを無くすと0.04までいくのですが、VisionSimulatorありでBodyROSなしだと0.8まで下がります。VisionSimulatorあり、BodyROSありが0.85まで上がったので、このあたりが限界かなとも思っています。

fkanehiro, 2015-10-27:
そうですね。このあたりがいいところでしょうね。
ところで今回の対応でトピックはシミュレーションのループとは完全に非同期になったわけですが、同期的に実行したい処理はＧａｚｅｂｏの場合はどうしているんでしょうね？

yosuke, 2015-10-28:
同期的に実行したいものはROS全体の場合もGazeboの場合もros_controlにすべて押し込んでしまうのが流儀みたいですね。PR2の場合はpr2_mechanismがその役目、atlasにもcontrol用パッケージをBostonDynamicsが出している（検索しても言及しか見当たらなかったのでクローズドソース？）ようです。

pub/subメッセージになってしまうとすべて非同期前提なので、やはりプラグインから同期して呼び出されるros_controlとの接続部分は必要そうですね。

yosuke, 2015-10-28:
この部分はRTMでいうところのrtcdっぽくなっているみたいですね。
http://wiki.ros.org/pr2_controller_manager