fkanehiro/choreonoid_ros_pkg #11: レンジデータを表示できない

fkanehiro, 2015-10-28:
rvizでレンジデータを表示しようとすると、次のような警告がでます。どうすればよいのでしょうか？
```
[ WARN] [1446025707.493125293]: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
```

yosuke, 2015-10-29:
rvizでレンジデータを表示するにはtfという関節＆リンクをツリー構造にしたデータを構築する必要があります（順運動学をたどったうえで画面に表示される）。
この部分、launchfileできれいに書きたいと思いますので、もう少々お待ちください。

yosuke, 2015-11-02:
rvizで表示する環境とtfの構築は完了しました。

joint_stateについては問題なくpublishできていることがrvizの画面から確認できました。

あとは各センサーデータのヘッダ部分にモデルのリンク情報を設定する必要があります（以下メモです）。

カメラは
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_camera_utils.cpp#L578

RGBDは
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_camera_utils.cpp#L578

レンジは
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_camera_utils.cpp#L578

imuは
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_camera_utils.cpp#L578

yosuke, 2015-11-02:
現状ここまで来ています。レーザースキャンデータにintensityという項目があり、おそらくそこに適切なデータを入れればスキャンデータも表示できると思うのですが、まだ値がよくわかっていません。

[image]

yosuke, 2015-11-03:
スキャンデータについても表示できるようになりました。RGBDの方もチェックします。

[image]

yosuke, 2015-11-04:
RGBDについても表示されました（間違えていた部分があったのでデバッグしました）。

[image]

closed 2015-12-25 by fkanehiro