fkanehiro/choreonoid_ros_pkg #22: pause_physics, unpause_physics トピックによる操作と Choreonoid による操作との整合性がとれていません

hmatsushita, 2015-11-30:
rostopic コマンドで /AISTSimulator/pause_physics による一時停止を実行した後、Choreonoid 側で一時停止のボタンを押してもシミュレーションが再開されません。
また、Choreonoid 側で一時停止を実施した後、rostopic コマンドで /AISTSimulator/unpause_physics を実行してもシミュレーションが再開されません。

fkanehiro, 2016-02-29:
SimulationBar::pauseSimulation()等を呼び出すべきかと思います。
https://github.com/s-nakaoka/choreonoid/blob/master/src/BodyPlugin/SimulationBar.h#L30
またGUI絡みの処理はGUIのイベントループから呼び出す必要があるので、callLaterを使うのだと思います。
https://github.com/s-nakaoka/choreonoid/blob/master/src/BodyPlugin/SimulationBar.h#L30
@hmatsushita さん、試してみてください。

hmatsushita, 2016-02-29:
情報ありがとうございます。
頂いた情報を元に修正を実施します。

hmatsushita, 2016-03-23:
SimulationBar クラスのメソッド呼び出しを試行しましたが、現行の SimulationBar クラスの実装では対応できないのではないかと考えています。

(恐らく) 当該クラスのパブリックメソッドは、GUI ボタンのイベントから呼び出す事を前提とした実装となっており、外部から呼び出して GUI の操作をする事は考慮されていないと思われます。

例えば、SimulationBar クラスの pauseSimulation の実装の概要は以下のようになっております (タイムバーの操作も実装されておりますが、こちらについては割愛しています)。

1.  一時停止ボタンが押された状態である場合 2. へ、そうでない場合は 3. へ

1.  シミュレーションが実行中の場合、シミュレーションを一時停止しメソッドを抜ける、シミュレーションが停止している場合は何もせずメソッドを抜ける

1.  シミュレーションが実行中の場合、シミュレーションをリスタートしメソッドを抜ける、シミュレーションが停止している場合は何もせずメソッドを抜ける

一時停止ボタンの状態を操作 (pauseToggle->setChecked(bool)) した上で pauseSimulation を呼び出せば、期待した動作になると思われますが、 pauseToggle はプライベートとなっているため、外部から操作する事ができません。

以上の事から、対応できないのではないかと考えた次第となります。

fkanehiro, 2016-03-23:
@s-nakaoka さん、何かコメントありませんでしょうか。