fkanehiro/choreonoid_ros_pkg #26: /[robotname]/set_joint_trajectory に誤ったパラメータを与えると Choreonoid がクラッシュする場合があります

hmatsushita, 2015-11-30:
Choreonoid ROS プラグインチュートリアルのプロジェクト (jvrc-1-rviz.launch) を開き、以下の誤ったメッセージ (第 3 引数がリストでない) を送信すると Choreonoid が Segmentaion fault でクラッシュします。

```
rostopic pub -1 /JVRC_1/set_joint_trajectory trajectory_msgs/JointTrajectory -- '{ seq: 0, stamp: 0, frame_id: ""}' '["NECK_Y"]' '[ positions: [0.2618], time_from_start: { secs: 0, nsecs: 0 } ]'
```

正しい (本来送信するつもりであった) メッセージは、以下となります。

```
$ rostopic pub -1 /JVRC_1/set_joint_trajectory trajectory_msgs/JointTrajectory -- '{ seq: 0, stamp: 0, frame_id: ""}' '["NECK_Y"]' '[ { positions: [0.2618], time_from_start: { secs: 0, nsecs: 0 } } ]'
```

また、他のパターンとして、以下のメッセージでは Choreonoid はクラッシュしませんが、

```
rostopic pub -1 /JVRC_1/set_joint_trajectory trajectory_msgs/JointTrajectory -- '{ seq: 0, stamp: 0, frame_id: ""}' '["NECK_Y"]' '[ ]
```

以下のメッセージでは Choreonoid が Segmentaion fault でクラッシュします。

```
rostopic pub -1 /JVRC_1/set_joint_trajectory trajectory_msgs/JointTrajectory -- '{ seq: 0, stamp: 0, frame_id: ""}' '["NECK_Y"]' '[ {} ]'
```

yosuke, 2015-11-30:
ROSのメッセージシリアライザのレベルでエラーが起こっているように思うのですが、Gazeboの場合はこのような壊れたメッセージを送ってもクラッシュはしないのでしょうか？

GazeboのROSプラグインコードも見ているのですが、ぱっと見そのような対処は入っていないようでした。

hmatsushita, 2015-12-25:
返答が大変遅くなってしまい申し訳ありません。
Gazebo での検証についてですが、trajectory_msgs/JointTrajectory を含む gazebo_msgs/SetJointTrajectory を提供するサンプルの用意が出来ず、現時点ではご質問への回答が満足に出来ない状態です。
重々、申し訳ありません。

取り急ぎ、deb パッケージで提供されている gazebo_ros の empty_world.launch に、URDF 形式に変換した JVRC-1 をスポーンし 、提供されるトピックやサービスへ壊れたメッセージを送信してみましたが、いずれもメッセージ送信前に以下のいずれかの状態となり、メッセージが送信されずクラッシュは発生しませんでした。

- 引数の指定が無い旨のエラーにより送信が実施されない

 コンソールに以下のようなメッセージが出力され送信が実施されません。

 ```
[WARN] [WallTime: 1451009833.187870] Inbound TCP/IP connection failed: field reference_frame must be of type str
 ```

- 引数の型が不一致である旨のエラーにより送信が実施されない

 コンソールに以下のようなメッセージが出力され送信が実施されません。

```
[WARN] [WallTime: 1451018658.334139] Inbound TCP/IP connection failed: <class 'struct.error'>: 'required argument is not a float' when writing 'link_name: JVRC-1_0_0::WAIST_Y_S
```

Choreonoid ROS package についても、本 issue の場合を除き、上記いずれかの状態となり、メッセージが送信される事はありませんでした。

今後、何か判明した時点で、速やかにコメントいたします。

hmatsushita, 2016-03-01:
21eba12 で修正しました。
callback に渡されたクラッシュするメッセージの内容を確認したところ、points 要素が 2 つ、先頭の要素にのみ positions が存在するメッセージとなっていました。
ROS 側でのチェックで引っかかった場合、メッセージが送信される事はありませんが、チェックを抜けてしまった場合、意図しないメッセージが届いてしまうので、受信側で対応する必要があるようです。

closed 2016-03-01 by hmatsushita