fkanehiro/choreonoid_ros_pkg #58: 関節軌道データの実行開始時間に誤りがあります

hmatsushita, 2016-06-20:
関節軌道データの実行開始時間が、関節軌道データ (trajectory_msgs::JointTrajectory) の timestamp + 先頭の軌道データ (trajectory_msgs/JointTrajectoryPoint[]) の duration ではなく、関節軌道データの timestamp で開始されています。

fkanehiro, 2016-06-20:
正しい仕様の記述はどこかにありますでしょうか？

hmatsushita, 2016-06-20:
サンプル等を含め正式な仕様の確認をいたします。
少々お時間をください。

hmatsushita, 2016-06-20:
http://wiki.ros.org/pr2_controllers/Tutorials/Moving%20the%20arm%20using%20the%20Joint%20Trajectory%20Action の「Time management of the trajectory.」に、ヘッダのタイムスタンプ  + 軌道データの time_from_start (duration) が指定関節角度の目標時間とあるので、timestamp で開始するのは誤りであると考えます。

fkanehiro, 2016-06-20:
ありがとうございます。
合わせて間違いと思われる箇所も明示して頂けると助かります。

hmatsushita, 2016-06-20:
https://github.com/fkanehiro/choreonoid_ros_pkg/blob/master/choreonoid_plugins/src/BodyRosJointControllerItem.cpp#L288 が該当箇所となります。

trajectory_start_ に ros::Duration(points_[0].time_from_start.sec, points_[0].time_from_start.nsec).toSec() を加算した時間が開始時間になると考えます。

あと、本件とは少し関係の無い話となりますが、ヘッダのタイムスタンプ指定が過去の時間であった場合、現在の実装ではリクエストを受信した時間を開始時間としていますが、この実装は正しいものでしょうか。
こちらでも調査いたしますが、一般的なロボットのコントローラとして、当該実装が正しいのか、もしくはこのような場合はリクエストを破棄するのが正しいのか、可能であればご返答いただければ幸いです。

fkanehiro, 2016-06-20:
上で示して頂いたURLの説明では、タイムスタンプの時刻から動き始め、最初の経由点のtime_from_start後にその経由点に到着するように動作するとあるので、示して頂いた行は特に問題ないと思いますがいかがでしょうか。
古いタイムスタンプの取り扱いについては、特に一般的な方法というのはないと思います。

hmatsushita, 2016-06-20:
ご指摘の通りです。
現状の実装が trajectory_start_ の時間 = 指定の関節角度への移動となっており、混同して誤認識しておりました。
お騒がせしてしまい申し訳ございません。
本 issue をクローズいたします。

また、古いタイムスタンプについてのご返答ありがとうございます。
こちらにつきましては、とりあえず現状の実装のままとしておきます。

closed 2016-06-20 by hmatsushita

fkanehiro, 2016-06-21:
タイムスタンプが現在の時刻よりも過去だった場合の対応については、gazebo_ros_pkgsを確認して頂けますでしょうか。

hmatsushita, 2016-06-21:
承知いたしました。