fkanehiro/choreonoid_ros_pkg #64: ビルドに失敗する (Choreonoid 325d8c3 使用)

hmatsushita, 2016-09-09:
catkin build choreonoid_ros_pkg 実行時、Choreonoid (325d8c3) のビルドに失敗します。

fkanehiro, 2016-09-09:
@hmatsushita さん、もうちょっと情報頂けないでしょうか。

@s-nakaoka さん、c++11対応絡みではないかと思いますが、何か注意点ありますか？(PPAやJenkinsのビルドも軒並み失敗しているようです）

hmatsushita, 2016-09-09:
すみません。
こちらで対応しようと調査しておりますが、C++11 絡みであるといった程度の情報しかありません。
何か分かり次第、対応もしくは情報の提出を行います。

fkanehiro, 2016-09-09:
どのように失敗するのか、でよいと思うのですが。

hmatsushita, 2016-09-09:
失礼しました。

In file included from /usr/include/c++/4.8/atomic:38:0,
                 from /home/mat/ros/indigo/catkin_ws_cnoidrospkg/build/choreonoid_ros/build/choreonoid-master/src/Util/Referenced.h:8,
                 from /home/mat/ros/indigo/catkin_ws_cnoidrospkg/build/choreonoid_ros/build/choreonoid-master/src/Util/Signal.h:8,
                 from /home/mat/ros/indigo/catkin_ws_cnoidrospkg/build/choreonoid_ros/build/choreonoid-master/src/Util/ConnectionSet.h:8,
                 from /home/mat/ros/indigo/catkin_ws_cnoidrospkg/build/choreonoid_ros/build/choreonoid-master/src/Util/ConnectionSet.cpp:5:

/usr/include/c++/4.8/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.

となっています。
コンパイルオプションを付ける事で対応できると考えております。

hmatsushita, 2016-09-09:
コンパイルオプションでビルドが先に進む事を確認しましたが、 Roki のサンプルビルドでビルドエラーが発生しました。

/home/mat/ros/indigo/catkin_ws_cnoidrospkg/build/choreonoid_ros/build/choreonoid-master/sample/Roki/Arm2dofController.cpp:8:17: error: ‘boost’ is not a namespace-name
 using namespace boost;
                 ^
/home/mat/ros/indigo/catkin_ws_cnoidrospkg/build/choreonoid_ros/build/choreonoid-master/sample/Roki/Arm2dofController.cpp:8:22: error: expected namespace-name before ‘;’ token
 using namespace boost;
                      ^
make[3]: *** [sample/Roki/CMakeFiles/Arm2dofController.dir/Arm2dofController.cpp.o] エラー 1

s-nakaoka, 2016-09-09:
ご迷惑をおかけしてすみません、Rokiについては私の方で動くよう修正します。

hmatsushita, 2016-09-09:
@s-nakaoka 様

いえ、とんでもございません。
対応のほどよろしくお願いいたします。

hmatsushita, 2016-09-09:
付けたコンパイルオプションの報告が漏れておりました。
-std=c++11 を使用しております。

Choreonoid のビルドは完了しましたが、プラグインのビルドに失敗するので、現在その対応を実施しております。

fkanehiro, 2016-09-09:
そのオプションは本来はCMakeLists.txtで指定しているので勝手につくはずです。それがつかないのはCMakeLists.txtの書き方か、ビルドの仕方に問題があるのではと思います。

hmatsushita, 2016-09-09:
情報ありがとうございます。
今一度見なおします。

s-nakaoka, 2016-09-09:
> Choreonoid のビルドは完了しましたが、プラグインのビルドに失敗するので、現在その対応を実施しております。

C++11対応にあたっては、主にstdで同様のものがあるboostのライブラリをstdのものに置き換える作業をしました。
プラグインについても、そこを意識して修正すればコンパイルできるかと思います。

具体的には、bind, function, shared_ptr 等のライブラリについて

・boostのものを使う場合は、boostの名前空間を省略せず、boost::bind, boost::function といったように書く。using namespace boost もしないほうがよいかも。
・上記についてChoreonoid本体のものがstdのものに置き換えられているので、なるべくstdのものに移行する。
・bindについては、 _1, _2 といったプレースホルダがstd版だとstd::placeholdersという名前空間に入るようになった。これを毎回つけるのは大変なので、using namespace std::placeholders; などとする。ただしboostのものは名前空間に入っておらず、混ざるとおかしくなる。まだboostのbindをインクルードしている外部ライブラリもあるので（PCL等）、混ざる場合は名前空間を省略せず書くか、namespace stdph = std::placeholders; などとして短くして書く。
・それか、bindはC++11のラムダ関数と置き換えることができるので、そちらを使う。その方がコールバックが分かりやすく書けてよいのでおすすめ。

といったようなところでしょうか。

fkanehiro, 2016-09-09:
@s-nakaoka さん、ちょっと誤解があるようなのですが、choreonoid本体がなぜかc++11のオプションがつかずビルドできないのです。

hmatsushita, 2016-09-09:
@s-nakaoka 様
情報ありがとうございます。
頂いた情報を元に対応を実施いたします。

s-nakaoka, 2016-09-09:
> @s-nakaoka さん、ちょっと誤解があるようなのですが、choreonoid本体がなぜかc++11のオプションがつかずビルドできないのです。

あれれ、そうなんですか。

c++11オプションは、トップディレクトリのCMakeLists.txtで
`
if(UNIX)
  SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
  add_definitions("-pthread")
endif()
`
としています。
当初ここを

add_definitions("-std=c++11 -pthread")

としていたのですが、これだとC言語ファイルのコンパイルにもオプションがついてコンパイラにおこられるので、いまのようになっています。

ただ以下をみたところc++11を有効にするやり方はいろいろあるようで、cmakeのバージョンとかにもよるのかもしれません。
http://stackoverflow.com/questions/10851247/how-to-activate-c-11-in-cmake

s-nakaoka, 2016-09-09:
この目的のための CMAKE_CXX_STANDARD という変数もあるようですね。https://cmake.org/cmake/help/v3.1/variable/CMAKE_CXX_STANDARD.html#variable:CMAKE_CXX_STANDARD

こちらを使ったほうがよいのかもしれません。

いずれにせよ、確実にオプションを有効に出来て、かつちょっと古めのcmakeでサポートされている書き方を模索する必要がありそうですね。こちらでももう少し検討してみます。

s-nakaoka, 2016-09-09:
Choreonoid付属のRokiPluginについては確かに今のままでコンパイルできているようです。

hmatsushita, 2016-09-09:
choreonoid_ros プラグインのソースコードを修正し chorenoid_ros_pkg が動作するところまで確認できました。
現在、.travis.yaml の方の確認作業を実施しております。

hmatsushita, 2016-09-09:
.travis.yaml の方は手を加える必要は無いように思われます。

Choreonoid へのコンパイルオプションの付け方の問題を解決した後に PR を投げるのが良いでしょうか。
それとも現状のものを一旦 PR として投げるのが良いでしょうか。

hmatsushita, 2016-09-09:
現状のものを PR #65 で投げましたが Travis CI でエラーとなっています。

cc1plus: error: unrecognized command line option -std=c++11 がエラー原因ですので、g++ のバージョンによる問題かと思われます。

s-nakaoka, 2016-09-09:
gccだと、

-std=gnu++11

というオプションもあるようです。
https://cmake.org/cmake/help/v3.1/prop_tgt/CXX_STANDARD.html#prop_tgt:CXX_STANDARD

fkanehiro, 2016-09-09:
以下のような手順でビルドすると-std=c++11のオプションが付かないようです。(ubuntu14)
PPAやjenkins等でビルドに失敗するのはこの辺に原因がありそうです。

$ git clone https://github.com/s-nakaoka/choreonoid
$ cd choreonoid
$ mkdir build
$ cd build
$ cmake ..
$ make VERBOSE=1

上でcmakeの代わりにccmakeを使うと付くようです。

s-nakaoka, 2016-09-12:
> 以下のような手順でビルドすると-std=c++11のオプションが付かないようです。

報告ありがとうございます。
これで気がついたのですが、どうもc++11オプションを有効化する記述がCMakeLists.txtの中で早すぎる位置にあったようです。位置を変えたところ、こちらの環境ではcmakeコマンド一発で c++11オプションも有効になるようになりました。
https://github.com/s-nakaoka/choreonoid/commit/28e1e07bc45b67ec6ddf6b9e632b2d1b8a326790

これでお試しいただけますでしょうか。

fkanehiro, 2016-09-12:
ありがとうございます。Launchpadでは14.04, 15.10ではコンパイルできるようになったのですが、15.04だと以下のように失敗します。
```
cd "/«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/obj-x86_64-linux-gnu/src/Base" && /usr/bin/x86_64-linux-gnu-g++   -DBOOST_DISABLE_ASSERTS -DCnoidBase_EXPORTS -DEIGEN_NO_DEBUG -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_OPENGL_LIB -D__OSVERSION__=2 -D__linux__ -D__x86_64__ -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -D_FORTIFY_SOURCE=2  -std=c++11 -O3 -DNDEBUG   -fPIC -I/usr/include/python2.7 -I/usr/include/eigen3 -isystem /usr/include/qt4 -isystem /usr/include/qt4/QtOpenGL -isystem /usr/include/qt4/QtGui -isystem /usr/include/qt4/QtNetwork -isystem /usr/include/qt4/QtCore -I"/«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1" -I"/«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/include" -I"/«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/thirdparty/irrxml-1.2/src"    -pthread  -fvisibility=hidden -o CMakeFiles/CnoidBase.dir/ShaderPrograms.cpp.o -c "/«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Base/ShaderPrograms.cpp"
In file included from /usr/include/eigen3/Eigen/Core:256:0,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Util/EigenTypes.h:8,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Util/BoundingBox.h:9,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/include/cnoid/BoundingBox:1,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Util/SceneGraph.h:10,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/include/cnoid/SceneGraph:1,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Util/SceneRenderer.h:9,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/include/cnoid/SceneRenderer:1,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Base/GLSceneRenderer.h:9,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/include/cnoid/GLSceneRenderer:1,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Base/GL1SceneRenderer.h:9,
                 from /«BUILDDIR»/choreonoid-1.6.0+dfsg+20160912+674+19~ubuntu15.04.1/src/Base/GL1SceneRenderer.cpp:6:
/usr/include/eigen3/Eigen/src/Core/util/Memory.h: In instantiation of ‘void Eigen::aligned_allocator<T>::construct(Eigen::aligned_allocator<T>::pointer, const T&) [with T = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>; Eigen::aligned_allocator<T>::pointer = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>*]’:
/usr/include/c++/4.9/bits/alloc_traits.h:253:4:   required from ‘static std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> std::allocator_traits<_Alloc>::_S_construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>; _Args = {const Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>}; _Alloc = Eigen::aligned_allocator<std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u> >; std::_Require<typename std::allocator_traits<_Alloc>::__construct_helper<_Tp, _Args>::type> = void]’
/usr/include/c++/4.9/bits/alloc_traits.h:399:57:   required from ‘static decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) std::allocator_traits<_Alloc>::construct(_Alloc&, _Tp*, _Args&& ...) [with _Tp = std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u>; _Args = {const Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>}; _Alloc = Eigen::aligned_allocator<std::_Sp_counted_ptr_inplace<{anonymous}::TransparentShapeInfo, Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>, (__gnu_cxx::_Lock_policy)2u> >; decltype (_S_construct(__a, __p, (forward<_Args>)(std::allocator_traits::construct::__args)...)) = <type error>]’
/usr/include/c++/4.9/bits/shared_ptr_base.h:620:37:   required from ‘std::__shared_count<_Lp>::__shared_count(std::_Sp_make_shared_tag, _Tp*, const _Alloc&, _Args&& ...) [with _Tp = {anonymous}::TransparentShapeInfo; _Alloc = Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>; _Args = {}; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]’
/usr/include/c++/4.9/bits/shared_ptr_base.h:1090:35:   required from ‘std::__shared_ptr<_Tp, _Lp>::__shared_ptr(std::_Sp_make_shared_tag, const _Alloc&, _Args&& ...) [with _Alloc = Eigen::aligned_allocator<{anonymous}::TransparentShapeInfo>; _Args = {}; _Tp = {anonymous}::TransparentShapeInfo; __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]’
/usr/include/c++/4.9/bits/shared_ptr.h:316:64:   required from ‘std::shared_ptr<_Tp>::shared_ptr(std::_Sp_make_shared_tag, const _Alloc&, _Args&& ...) [with _Alloc = Eigen::aligned_allocator<{anonymous}: [... cut ...]
/usr/include/c++/4.9/bits/shared_ptr_base.h:494:11: error: use of deleted function ‘std::_Sp_counted_base<_Lp>::_Sp_counted_base(const std::_Sp_counted_base<_Lp>&) [with __gnu_cxx::_Lock_policy _Lp = (__gnu_cxx::_Lock_policy)2u]’
/usr/include/c++/4.9/bits/shared_ptr_base.h:203:7: note: declared here
       _Sp_counted_base(_Sp_counted_base const&) = delete;
       ^
make[3]: *** [src/Base/CMakeFiles/CnoidBase.dir/GL1SceneRenderer.cpp.o] Error 1
src/Base/CMakeFiles/CnoidBase.dir/build.make:1907: recipe for target 'src/Base/CMakeFiles/CnoidBase.dir/GL1SceneRenderer.cpp.o' failed
```

s-nakaoka, 2016-09-13:
エラーメッセージをみてみたのですが、これだけではどう直してよいかちょっと分からないですね・・・。Eigenのalignment指定を含む型とshared_ptrを組み合わせることで出ているようにも見えますが、他の環境ではコンパイルできていますので、Eigenのこのバージョンに不備があるのかもしれません。

もしそうだとして、Ubuntu15.04自体は既にサポートが切れていてもうアップデートもできないと思うので、今から15.04の環境を構築して更なる調査をするというのも、労多くして益少なしということで、できれば避けたいと思います。

fkanehiro, 2016-09-13:
了解しました。15.04もビルド対象から外します。
HPに明記されているのは、14.04, 16.04ですので公式サポートとしてはこれら2つになるのでしょうか？
18.04がでたら16.04と18.04になる？

s-nakaoka, 2016-09-13:
> HPに明記されているのは、14.04, 16.04ですので公式サポートとしてはこれら2つになるのでしょうか？

はい、そのようにしたいと思います。

> 18.04がでたら16.04と18.04になる？

はい、18.04が出たらそれもサポートしたいと思います。
３つ前のLTSバージョンについては、可能であればサポート期間中はサポートしたいとは思います。そういう意味では12.04もまだ少しサポート期間が残っているのですが、C++11対応が難しそうなので今回は断念しました。そのあたりは状況を見ながらということになるかと思います。

fkanehiro, 2016-09-13:
了解です。対応ありがとうございました。

closed 2016-09-13 by fkanehiro