fkanehiro/choreonoid_ros_pkg #83: シミュレーション実行中に動的ロードしたモデルの物理接触が発生しない

hmatsushita, 2017-03-02:
シミュレーション実行中に /choreonoid/spawn_*_model サービスを使用し動的にモデルをロードした場合、ロードしたモデルの物理接触が発生しません。
移動中のロボットの進行方向にモデルをロードし、ロボットがロードしたモデルをすり抜ける事を確認しました。
また、以下の内容について確認しました。

* /choreonoid/[worldname]/physics/contacts で接触発生が確認できない事

* /choreonoid/links_states や /choreonoid/model_states ではロードしたモデルの存在が確認できる事

* Choreonoid の GUI 側 (ボディ／リンクビューの「干渉」表示) では接触発生が確認できる事

以上の事から、物理エンジンへロードしたモデルの通知・登録に類する処理がなされていないように見受けられます。

fkanehiro, 2017-03-02:
はい、残念ながら現状の構造ではシミュレーション中に物体を追加するということはできないと思います。

hmatsushita, 2017-03-02:
承知いたしました。
マニュアルの方へは機能制約として追記いたします。

fkanehiro, 2017-03-02:
ちなみにシミュレーション開始前に追加するとどうでしょうか。

hmatsushita, 2017-03-02:
シミュレーション停止時はサービスが動作しておりませんので追加が出来ませんが、シミュレーション実行中に追加しシミュレーションを停止、再度シミュレーションを開始した場合につきましては、問題なく接触が発生いたします。