fkanehiro/choreonoid_ros_pkg #85: buildに失敗する

wt-nsakai, 2017-09-18:
現在，Choreonoid ROS Pluginのインストールを試みておりますが，
下記サイトのインストール手順に沿って行ったところ，以下のエラーが
発生し，ビルドに失敗しております．
＜参考サイトURL＞
https://fkanehiro.github.io/choreonoid_ros_pkg_doc/html-ja/install.html

エラーが発生しているのは，sdfloaderをビルドするタイミングで，
"catkin build choreonoid_sdfloader_plugin"
というコマンドを実行したところです．

以下のファイルが見つからないというエラーが出力されたので，
既にインストールをし，動作が確認されているPCから該当するファイルを
コピー＆ペーストをしたところ，このエラーは解消されました．
<cnoid/DaeParser>
"src/Util/DaeParser.h"
<cnoid/DaeNode>
"DaeNode.h"

上記のエラーを解消した後，再度buildを行ったところ，
添付の画像のようなエラーが出力されて，buildが完了しませんでした．
エラー部分の抜粋を以下に示します．

```
Errors     << choreonoid_sdfloader_plugin:make /home/biped/catkin_ws/logs/choreonoid_sdfloader_plugin/build.make.005.log               
/home/biped/catkin_ws/src/choreonoid_sdfloader_plugin/SDFBodyLoader.cpp: In member function ‘bool cnoid::SDFBodyLoaderImpl::load(cnoid::Body*, const string&)’:
/home/biped/catkin_ws/src/choreonoid_sdfloader_plugin/SDFBodyLoader.cpp:590:28: error: passing ‘const Position {aka const Eigen::Transform<double, 3, 18>}’ as ‘this’ argument of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>& Eigen::Transform<Scalar, Dim, Mode, _Options>::operator=(const Eigen::Transform<Scalar, Dim, Mode, _Options>&) [with _Scalar = double; int _Dim = 3; int _Mode = 18; int _Options = 0]’ discards qualifiers [-fpermissive]
                 link->Tb() = rootpose * (*c)->pose;
                            ^
make[2]: *** [CMakeFiles/CnoidSDFLoaderPlugin.dir/SDFBodyLoader.cpp.o] エラー 1
make[1]: *** [CMakeFiles/CnoidSDFLoaderPlugin.dir/all] エラー 2
make: *** [all] エラー 2
cd /home/biped/catkin_ws/build/choreonoid_sdfloader_plugin; catkin build --get-env choreonoid_sdfloader_plugin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.......................................................................................................................................
Failed     << choreonoid_sdfloader_plugin:make           [ Exited with code 2 ]                                                        
Failed    <<< choreonoid_sdfloader_plugin                [ 4.9 seconds ]
```

このエラーの解消方法について，ご存知の方がいらっしゃいましたらご教示いただきく思います．
また，ファイルが見つからないという問題の解消についてもご検討いただけますと幸いです．

以上，よろしくお願いいたします．

fkanehiro, 2017-09-19:
ご指摘ありがとうございます。
Choreonoid本体のＡＰＩに変更があったようですので修正しました。
https://github.com/fkanehiro/choreonoid-sdfloader-plugin/commit/d5e86f37cdff1a57d8862365f6adeb48fa20c786

wt-nsakai, 2017-09-20:
本日改めてbuildを行ったところ，エラーもなく終了いたしました．
対応いただき，ありがとうございました．

closed 2017-09-20 by wt-nsakai