fkanehiro/choreonoid_ros_pkg #86: choreonoid_ros_pkgがビルドできない

wt-nsakai, 2017-09-25:
現在下記のサイトに沿ってChoreonoid ROS のインストールを
試みておりますが，choreonoid_ros_pkgのビルドに失敗しております．
＜参考サイト＞
https://fkanehiro.github.io/choreonoid_ros_pkg_doc/html-ja/install.html

具体的には
`catkin build choreonoid_ros_pkg`
を実行したところで以下のようなエラーが出力されます．

```
Errors     << choreonoid_plugins:make /home/biped/catkin_ws/logs/choreonoid_plugins/build.make.000.log                                                     
In file included from /home/biped/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/Util/pybind11/PyUtil.h:9:0,
                 from /home/biped/catkin_ws/devel/include/choreonoid-1.6/cnoid/PyUtil:3,
                 from /home/biped/catkin_ws/src/choreonoid_ros_pkg/choreonoid_plugins/src/python/PyRosPluginModule.cpp:7:
/home/biped/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/Util/pybind11/PyReferenced.h:5:31: fatal error: pybind11/pybind11.h: そのようなファイルやディレクトリはありません
 #include <pybind11/pybind11.h>
                               ^
compilation terminated.
In file included from /home/biped/catkin_ws/src/choreonoid_ros_pkg/choreonoid_plugins/src/python/PyItems.cpp:10:0:
/home/biped/catkin_ws/devel/include/choreonoid-1.6/cnoid/PyBase:3:38: fatal error: src/Base/pybind11/PyBase.h: そのようなファイルやディレクトリはありません
 #include "src/Base/pybind11/PyBase.h"
                                      ^
compilation terminated.
make[2]: *** [src/python/CMakeFiles/PyRosPlugin.dir/PyRosPluginModule.cpp.o] エラー 1
make[2]: *** 未完了のジョブを待っています....
make[2]: *** [src/python/CMakeFiles/PyRosPlugin.dir/PyItems.cpp.o] エラー 1
make[1]: *** [src/python/CMakeFiles/PyRosPlugin.dir/all] エラー 2
make: *** [all] エラー 2

```
エラー文に書かかれている場所を確認すると，確かに当該のファイルはありません．
また，以前の環境では生成されないファイルだったようですが，環境に変更があったのでしょうか．

以上，お手数ですが，ご確認いただけますと幸いです．

fkanehiro, 2017-09-25:
Choreonoid本体の変更に追従させました（ ce95d79056c7016ff528d3ba49bf74b2f383bd0a ）

wt-nsakai, 2017-09-27:
無事ビルドができました．
ご対応いただき，ありがとうございました．

closed 2017-09-27 by wt-nsakai