fkanehiro/choreonoid_ros_pkg #87: ビルドに失敗します

m-asahi, 2017-11-17:
当方、Ubuntu14.04, gcc 5.4.1, eigen 3.3.4でchoreonoid_ros_pkgのビルドを行ったところ、
catkin build choreonoid_ros_pkg でchoreonoidのビルドがエラーとなってしまいます。

In file included from /home/asahi/catkin_ws/src/choreonoid_ros_pkg/choreonoid_plugins/src/BodyRosJointControllerItem.cpp:6:0:
/home/asahi/catkin_ws/src/choreonoid_ros_pkg/choreonoid_plugins/src/BodyRosJointControllerItem.h:49:24: error: ‘Target’ has not been declared
     virtual bool start(Target* target);

Targetについては以下のようなメッセージも出ています。

/home/asahi/catkin_ws/src/choreonoid_ros_pkg/choreonoid_plugins/src/BodyRosItem.cpp:68:25: note: suggested alternative:
In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/Util/EigenTypes.h:8,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/EigenTypes:1,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/Body/Link.h:10,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/Body/Body.h:10,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/Body/CollisionLinkPair.h:9,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/CollisionLinkPair:1,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/BodyPlugin/CollisionSeq.h:9,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/BodyPlugin/SimulatorItem.h:9,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/src/BodyPlugin/ControllerItem.h:9,
                 from /home/asahi/catkin_ws/devel/include/choreonoid-1.6/cnoid/ControllerItem:1,
                 from /home/asahi/catkin_ws/src/choreonoid_ros_pkg/choreonoid_plugins/src/BodyRosItem.h:4,
                 from /home/asahi/catkin_ws/src/choreonoid_ros_pkg/choreonoid_plugins/src/BodyRosItem.cpp:1:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:468:5: note:   ‘Target’
     Target = SSE
     ^
これはEigenのバージョンによるものでしょうか？

fkanehiro, 2017-11-26:
eigenもgccも標準とは異なるバージョンを入れられているのですね。
同じ環境を用意できないので確認できないのですが、ご推察の通りEigenやコンパイラのバージョンによるものかもしれません。あるいはchoreonoid本体の変更によるものかもしれませんのでこちらは確認してみます。

fkanehiro, 2017-11-27:
確認したところ、choreonoid本体の変更に伴ってコンパイルができなくなったようです。
対応を検討しますのでしばらくお待ち下さい。

fkanehiro, 2017-11-27:
s-nakaoka/choreonoid@0d26d0006fd6f7a601fa162bccb71d94ff8dd779
で互換性を維持するよう変更が行われました。
また何か問題ありましたらお知らせください。

closed 2017-11-27 by fkanehiro

m-asahi, 2017-11-28:
無事ビルドできました！ ありがとうございます。

ただ、catkin buildそのままでは通らず、BodyRosItem::updateRangeSensorにある、sensor->pitchResolution/yawResolutionを、それぞれpitchStep/yawStepに直す必要がありました。choreonoid本体の最近の変更と食い違ってしまったようです。

fkanehiro, 2017-11-30:
ご指摘ありがとうございます。
2896cf570800610e6931c74f6aa79cdc0f6ddd6b で修正しておきました。