fkanehiro/choreonoid_ros_pkg #88: jvrc-1-rviz.launchでChoreonoidが落ちます

m-asahi, 2017-12-01:
おかげさまで無事ビルドが通り、 roslaunch choreonoid_ros jvrc-1-rviz.launch を実行しましたが、Choreonoidが強制終了と再起動を繰り返してしまい動作しません。

環境は#84のときと同じですが、GPUにNVIDIA GeForce GTX 1060を使用しています。
ドライバはNVIDIA binary driver version 384.98（aptパッケージで言うnvidia-384）です。
オープンソース版ドライバでGLVisionSimulatorを使用したシミュレーションを実行すると、Choreonoidが異常終了するとありましたが、ここでいうオープンソースドライバはnouveauドライバですよね？

実際にroslaunch choreonoid_ros choreonoid.launch で空のプロジェクトを立ち上げ、SR1とSR1LiftupController, CameraSampleControllerを組み合わせて、GLVisionSimulatorでカメラ画像を取得することはできました。

なお、choreonoid_sdfloader_pluginは組み込んでいません。

fkanehiro, 2017-12-02:
ご報告ありがとうございます。
現象確認できましたので調査中です。
しばらくお待ちください。

fkanehiro, 2017-12-11:
お待たせいたしました。
choreonoid本体の変更に合わせて修正を行いました（ 84e7fe199c96c96bc9f75234617932f54bca86eb ）。
また何かありましたらお知らせください。

m-asahi, 2017-12-12:
無事に動作しました。
ありがとうございます。

closed 2017-12-12 by m-asahi