fkanehiro/choreonoid_ros_pkg #89: YAML形式モデルのカメラが配信されない

m-asahi, 2018-01-19:
棒の先にカメラがついた単純なモデルcamをyaml形式で作成して、choreonoidでボディとして読み込み、小アイテムとしてBodyRosItemをつけましたが、/cam/camera/image_rawが配信されません。（choreonoid内でシーンビューを見ることはできています）

[cam.body](https://github.com/fkanehiro/choreonoid_ros_pkg/files/1645721/cam.body.txt)

このcamモデルをサンプルのjvrc-1-rviz.launchの中に入れてみたところ、rcamera/lcameraのimage_rawは配信されますが、やはり/cam/camera/image_rawは配信されません。

BodyRosItem::createSensors の動きを追ってみたところ、メンバ変数visionSensors_について、JVRC-1では２つのカメラが読み込まれますが、camではこれが空のままなのがわかりました。

Body::devicesが空なのが、VRMLBodyLoaderとYAMLBodyLoaderの間の差異に起因するのだとすると、原因はChoreonoid本体側にあるかとも思われますが、ひとまずこちらでご報告します。

yosuke, 2018-09-27:
同様の問題を確認。

本家の実装を見るとデバイス取得APIの使い方が少し変わってしまっていそう。

https://github.com/s-nakaoka/choreonoid_rosplugin/blob/master/src/BodyPublisherItem.cpp#L265

yosuke, 2018-09-27:
BodyRosItemが実装された当初はgetSortedById()をコールするとすべてのセンサが取得できていたが、bodyの規格を作るにあたってIDが振られていないセンサの作成を許容したのかもしれない（その場合はgetSortedById()だと取得できないのかもしれない）。

yosuke, 2018-09-27:
無事に修正できたと思う。