fkanehiro/choreonoid_ros_pkg #95: PutPropertyFunctionの更新への対応

ssr-yuki, 2020-01-13:
2019/1/12の [Remove unused functions of PutPropertyFunction](https://github.com/s-nakaoka/choreonoid/commit/198fd947fa4f20ec67d183d5ce0dd25ce35ebf13#diff-9c19d4baa1415fc05e9b60a622acf61f) においてクラス`PutPropertyFunction`のメンバが変更され，現在`choreonoid_plugins`内で使用されいている`changeProperty`や`putProperty`といった関数が使えなくなっています．

別の手法で再実装するか，あるいは現行のpulginの形式を見直す必要があるかもしれません．

RyodoTanaka, 2020-01-14:
@ssr-yuki 
https://github.com/WRS-TNK/choreonoid_ros_pkg/tree/dev/latest-choreonoid
こちらのリポジトリで，`cnoid/PutPropertyFunction`をincludeすることで対応してみました．

fkanehiro, 2020-01-14:
ありがとうございます。PR投げて頂ければマージします。

RyodoTanaka, 2020-01-14:
@fkanehiro 
お返事ありがとうございます．
本件の変更ですが，Choreonoidのバージョンに依存しています．
Choreonoid-1.7までは変更不要で，
Choreonoid-1.8の一部から変更が必要なようです．

以上より，ブランチを切るかタグを付けて管理されたほうが良いかと思いますが，いかがでしょうか？
`choreonoid-1.7-dev`，`choreonoid-1.8-dev`等のブランチがあれば良いと思います．
よろしくお願いします．

fkanehiro, 2020-01-14:
CNOID_MINOR_VERSIONを使って場合分けするのは難しい感じでしょうか？

RyodoTanaka, 2020-01-14:
CNOID_MINOR_VERSIONを使えれば場合分けは可能と思います．
が，すみません．choreonoid_ros_pkg/choreonoid_plubins ではどの部分を変更すれば対応可能になりますか？

fkanehiro, 2020-01-14:
すみません、ご質問の意味がよくわからないのですが、
https://github.com/WRS-TNK/choreonoid_ros_pkg/commit/31ea604655406f1bf7ee1850ec41f09005e4b88a
で変更された部分がversion 1.8以降でのみ有効になるようにすればよいのではないでしょうか。

RyodoTanaka, 2020-01-14:
@fkanehiro 
あ，意味がわかりました．
すみません．
コンパイル時に場合分けするためにCMakeLists.txtを変更するものと勘違いしていました．
変更次第，PRを出します．
よろしくお願いします．

ssr-yuki, 2020-01-14:
@RyodoTanaka 
対応ありがとうございます、よろしくお願いします

s-nakaoka, 2020-01-15:
すみません、このオーバーロード関数についてはもう使われていないと思ってしまい、ご迷惑をおかけしました。もう理解されているかと思いますが、このオーバーロードはboolを返さない関数を指定しても最終的にboolでtrueを返す関数として処理するというものです。以前はboost::bindで関数オブジェクトを与えることが多かったのですが、C++11以降はラムダ式が使えるので、ラムダ式の中で最後に return true; を入れるだけで同じことができますので、もう必要なくなったと判断した次第です。しかしここに影響が及んでしまい申し訳ありませんでした。

s-nakaoka, 2020-01-15:
すみません、勘違いしました。changePropertyの方が影響したようですね。こちらもラムダ式があれば同じことを簡単に書けますが、安易に消してしまったのはよくなかったかもしれません。

RyodoTanaka, 2020-01-21:
本件，PR出すのが遅れていてすみません．
本件の対応以外にも修正を行わないといけない箇所があるので，それを精査＆分割してPR出すよう準備しています．
完了次第出しますので，よろしくお願いします．

ssr-yuki, 2020-02-09:
Pythonディレクトリ以外で発生しているエラーを全て除いて，PRを出させていただきました．
ご確認いただけると幸いです．

closed 2020-02-19 by ssr-yuki